TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin #6: Default pulse duration is actually 60ms (milliseconds) - not 60us (microseconds)(?)

ThePhilLee, 2015-10-27:
````
ppmValue = "L60000 P%.2f B1 D0" % altppm
````
This seems to be the line that assigns the pulse duration. Assuming the units for "L60000" is microseconds (10^-6), the default pulse duration is actually 60 milliseconds (10^-3). The readme says 60us, while the plugin help tab says 60ms.

I burned some lines at feed=5000 mm/min ppm=0.075 and default L60000 as a test to see how long the actual pulses were. It resulted in dashed lines, and each dash has a width of 4.6mm-5.2mm. Unless my math is wrong, at 5000 mm/min, a pulse of 60 milliseconds would result in a line width of 5 mm. A pulse of 60 microseconds, on the other hand, would result in a width of .005 mm.

On a related note, it would be great if we could assign the pulse duration in each layer as we do the feed and ppm. Or at the very least, if we can assign a default pulse duration. Great plugin by the way.

TurnkeyTyranny, 2015-10-29:
Hey nice find! I think you're right, the units are probably typo'd. I'll check this out.

dreydean, 2015-12-25:
Any news on this?