TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin #7: Full Circle Cut, Need Option To Close Gap. (Over Lap)

xboxhacker, 2015-11-24:
Working on cutting 3/16" acrylic sheet. When I cut a full circle, I need the cut to go a little further to "close the gap" or overlap the start/end point.

Example: Cut circle with a circumference of 100mm. I need the cut to go 100.05mm in circumference  to overlap the start/end points.

The exporter is working 100% fine. I want to know if this option can be added.

I am running Smoothieware on a MKS Sbase board. Cutting through Pronterface on Ethernet.

Attached is a 10X photo on the start/end point.

[image]

Thank you!

Cromaglious, 2016-07-21:
I've run into this problem myself...  Now researching a solution.

Cromaglious, 2016-07-21:
a lead in and leadout ala plasma cutter would be nice too...  or put a line tangential to the start stop point on the out side of a circle.  Haven't figured out on for an inside corner.