TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin #11: Feed speed on raster G0 not picked up

Cromaglious, 2016-05-26:
somewhere around line 635 in turnkeylaser.py

     if(cutFeed < self.options.Mfeed):
            gcode += 'G0 X'+str(curve['x'])+' Y'+str(curve['y'])+' F'+str(self.options.Mfeed)+'\n'
        gcode += 'G0 X'+str(curve['x'])+' Y'+str(curve['y'])+' '+cutFeed+'\n'

basically compares str F200 < int 2200
I changed mine to 

     if(cutFeed != "F"+str(self.options.Mfeed)):
            gcode += 'G0 X'+str(curve['x'])+' Y'+str(curve['y'])+' F'+str(self.options.Mfeed)+'\n'
        gcode += 'G0 X'+str(curve['x'])+' Y'+str(curve['y'])+' '+cutFeed+'\n'

nextime, 2016-05-29:
I've changed it in my fork with:
if(int("".join([x for x in list(cutFeed) if x.isdigit()])) < int(self.options.Mfeed)):

Cromaglious, 2016-07-21:
nextime,

Thanks that's a better Solution, I'm changing mine as well to;

if(int("".join([x for x in list(cutFeed) if x.isdigit()])) < int(self.options.Mfeed)):