TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin #14: Nonsense G-Code output file when running against the cutting_surface.svg

dale6998, 2016-08-15:
Hello,

I'm running the plugin against the included `cutting_surfaces.svg` and am getting an outputted g-code file, `output.g`.  This G-code is not correct (I don't think).

`; Raster data will always precede vector data           
; Default Cut Feedrate 300 mm per minute
; Default Move Feedrate 2000 mm per minute
; Default Laser Intensity 10 percent
G28 XY; home X and Y

M5 ;turn the laser off

;(************************************************************)
;(***** Layer: 10 [ppm=30,feed=2500]                     *****)
;(***** Laser Power: 10                                  *****)
;(***** Feed Rate: 2500.0                                *****)
;(***** Pulse Rate: 30.0                                 *****)
;(************************************************************)
;(MSG,Starting layer '10 [ppm=30,feed=2500]')

;Beginning of Raster Image image4556 pixel size: 753x750
M649 S10 B2 D0 R0.09406
G0 X11.518 Y15.927 F2500

G7 $1 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAMHCiA3Tl9xg42XoqOlp6CZkoZ5bFpINSQTAQEAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAhMkNUda
G7 L28 DbHmGkpmgp6WjopiNg3FfTjchCwcD

G7 $0 L68 DBQoQMVN1kKvG1eT09vn78efcybajiGxQNx0DAgEAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQIDHTdRbIikt8rd59H89fb05NXGq5B2
G7 L56 DUzEQCgUAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=

G7 $1 L68 DAQMFFCQ0MzEwIBAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACRIcMERYcImitMbY4u339fv8
G7 L68 D9u9o287Br5yKdV9KOSkYEAgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAACBAYKTpKX3SJnK9Bztvn7vX899n37ePYxrSiiXFYRDAcEgkAAAAAAAAA
G7 L36 DAAAAAAAAAAAAAAAAAAAAECAwMjM0JRUFAwE=

G7 $0 L68 DAwcMK0tqZ2ViQSAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAEiU5W39iscHQ2eLr8fb79P39
G7 L68 D9vbz7ebg183EtKOTc1IyIRAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAECExUnOTo7TEzdfg5u3z99r99fz79vDr4tnQwbGif1s5JRIAAAAAAAAA
G7 L36 DAAAAAAAAAAAAAAAAAAAAIUJiZGdqSioLBwM=

G7 $1 L68 DBQsRP29gnJiUYzIBAAAAAAAAAAAAAAAAAAAAAAAAAAAAHDlWh7rt8vj9999999999999
G7 L68 D99999999999989fcrHtLMhkAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
G7 L68 DAAAAAAAAAAAAGDJLe6zb6PP99999999999999999999999999PLsuodWORwAAAAAAAAA
G7 L36 DAAAAAAAAAAAAAAAAAAAAMWOTmJyfcUESCwUA

`

When I compare this to the included g-code for, say the `wolf.gcode`, I see something much more reasonable:

`G28 ; home all

M5

;(************************************************************)
;(***** LAYER: 10                                        *****)
;(************************************************************)
;(MSG,Starting layer '10')

G00 X3.299782 Y118.552030
T01 (select tool)
M6 (tool change)

M5

G00 X3.299782 Y118.552030 F1400
M3
G02 S10.00 X3.548178 Y118.501119 I0.000000 J-0.631423 F1400
G02 S10.00 X3.676074 Y118.391017 I-0.110031 J-0.257148 F1400
G03 S10.00 X3.805945 Y118.305855 I0.156134 J0.096495 F1400
G03 S10.00 X4.136655 Y118.324427 I0.119469 J0.826362 F1400
G03 S10.00 X4.290573 Y118.498593 I-0.052748 J0.201711 F1400
G03 S10.00 X4.208141 Y118.860798 I-0.527850 J0.070353 F1400
G02 S10.00 X3.894311 Y120.088890 I1.636677 J1.072386 F1400
G02 S10.00 X4.381019 Y121.026364 I1.354478 J-0.108126 F1400
G03 S10.00 X4.518115 Y121.254128 I-0.257294 J0.310012 F1400
G03 S10.00 X4.521857 Y121.701707 I-1.117987 J0.233151 F1400
G01 S10.00 X4.429717 Y122.183697 F1400
G01 S10.00 X4.895852 Y121.878273 F1400
G03 S10.00 X5.715594 Y121.570679 I1.094221 J1.670000 F1400
G03 S10.00 X5.907575 Y121.737854 I0.023203 J0.167175 F1400
G02 S10.00 X5.950319 Y121.885151 I0.275167 J0.000000 F1400
G02 S10.00 X6.193082 Y122.188362 I1.490701 J-0.944727 F1400
G03 S10.00 X6.337743 Y122.501879 I-0.353021 J0.353021 F1400
G03 S10.00 X6.258010 Y123.006380 I-1.089013 J0.086441 F1400
G02 S10.00 X6.117066 Y123.526356 I1.949547 J0.807532 F1400
G02 S10.00 X6.177659 Y123.620719 I0.085465 J0.011756 F1400
G02 S10.00 X6.324337 Y123.604248 I0.054090 J-0.179649 F1400
G02 S10.00 X6.917546 Y123.204954 I-2.313533 J-4.077384 F1400
G03 S10.00 X7.645279 Y122.813157 I1.472437 J1.863187 F1400
G03 S10.00 X8.523666 Y122.674535 I0.862104 J2.610473 F1400
G02 S10.00 X9.426753 Y122.577076 I0.023837 J-4.012027 F1400
G02 S10.00 X10.172822 Y122.304135 I-0.672670 J-2.994842 F1400
G03 S10.00 X10.913331 Y122.007204 I1.862945 J3.574118 F1400
G03 S10.00 X11.469203 Y121.928241 I0.555871 J1.917087 F1400
G02 S10.00 X12.014373 Y121.830590 I-0.000000 J-1.570622 F1400
G02 S10.00 X12.732505 Y121.460672 I-1.159757 J-3.133495 F1400
G03 S10.00 X13.106806 Y121.272040 I0.908357 J1.336769 F1400
G03 S10.00 X13.169833 Y121.350525 I0.017640 J0.050384 F1400
G03 S10.00 X12.872712 Y121.779657 I-4.335280 J-2.684228 F1400
G03 S10.00 X11.214153 Y123.912956 I-264.217574 J-203.707714 F1400
G03 S10.00 X8.237985 Y127.411776 I-45.245505 J-35.471544 F1400
G03 S10.00 X7.805944 Y127.609683 I-0.452414 J-0.417104 F1400
G02 S10.00 X7.250586 Y127.803711 I0.032664 J0.985294 F1400
G02 S10.00 X6.125693 Y128.858360 I4.039718 J5.436012 F1400
G02 S10.00 X5.527842 Y129.862437 I2.688630 J2.280899 F1400
G02 S10.00 X5.048788 Y131.630257 I8.073594 J3.136650 F1400
G02 S10.00 X5.044642 Y133.608193 I6.261267 J1.002096 F1400
G02 S10.00 X5.641115 Y135.433244 I5.903813 J-0.919520 F1400
G03 S10.00 X5.856399 Y135.960974 I-2.818949 J1.457744 F1400
G03 S10.00 X5.907575 Y136.302575 I-1.114495 J0.341601 F1400
G02 S10.00 X6.140782 Y137.422636 I2.806357 J-0.000000 F1400
G02 S10.00 X7.146571 Y139.342969 I13.830231 J-6.020132 F1400
G02 S10.00 X8.431644 Y141.085998 I11.390870 J-7.052845 F1400
G02 S10.00 X9.092021 Y141.445693 I0.814092 J-0.708568 F1400
G02 S10.00 X9.500589 Y141.433281 I0.168827 J-1.173385 F1400
G02 S10.00 X10.506332 Y141.153635 I-1.563568 J-7.571749 F1400
G03 S10.00 X11.510442 Y140.820685 I9.460680 J26.850881 F1400
G03 S10.00 X11.999035 Y140.689270 I1.934711 J6.219171 F1400
G01 S10.00 X12.493093 Y140.576774 F1400
G01 S10.00 X12.021565 Y141.514414 F1400
G`

I'm running Arch 64bit, and I changed the first line of the .py file in the inkscape exensions folder to read:
`#!/usr/bin/env python2`, since Arch defaults to Python3 now.  Also, the PIL and lxml modules are all python2, installed via PACMAN.

```
pacman -S python2-lxml
pacman -S python2-imaging
```

Inkscape is 64-bit, but that shouldn't cause this....that should cause a total failure.

TurnkeyTyranny, 2017-03-29:
G7 is raster data, I made it up myself for my own firmware. It's base64 encoded data for power levels and pixels.

TurnkeyTyranny, 2017-03-29:
The firmware that understands G7 commands is also linked in the readme from this project.

closed 2017-03-29 by TurnkeyTyranny

mkeyno, 2017-06-06:
@dale6998 my plugin not created G7 code , do you know what did I miss?

dale6998, 2017-06-29:
What do you mean?  Not sure I understand what you're asking.

mkeyno, 2017-07-02:
@dale6998  I've tried to create G7 raster code with inkscape & TurnkeyTyranny plugin but every setting I choose , the extension just create the vector Gcode not the G7 raster code, somehow the pluging only interpreted my image as vector image

dale6998, 2017-07-25:
I'm importing all my stuff as a PDF from GIMP.  I'm actually outputting from a CAD program (KiCad) and importing into GIMP and then exporting as a 1200dpi PDF to bring into InkScape, since I can't use it otherwise.  I'd like to have your problem...