TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin #15: Exports gcode but then freezes

ExplodingLemur, 2016-08-22:
Inkscape 0.91, used both 32 and 64 bit versions on Windows, with Python 2.7.12 (also 32 and 64 bit versions).  I run the exporter, it creates the gcode file, but keeps the file handle open and the "Turnkey Laser Exporter working, please wait" dialog remains open in Inkscape.  If I kill the Inkscape process the gcode file handle closes and it has the correct code in it for my laser, but I'm not sure why it isn't finishing gracefully.

Cwoodmaker, 2017-01-21:
Im having exactly the same issue.  Did you ever solve ?

ExplodingLemur, 2017-01-21:
I have not.

wdowner, 2017-01-25:
I have the same problem as well, although it is not consistently occurring. It seems to occur more often when trying to export a raster image.

skallet, 2017-03-23:
Same issue. Inkspace 0.92, Python 2.7.10 .. seems very random but for same paths, it freezes every time.

mayhem2408, 2017-03-23:
I have not had any problems with the plugin keeping the file open. I have had problems with inkscape that when the please wait dialog is shown and inkscape looses focus, sometimes it is hard to get the dialog box back to click on OK and it give the appearance that inkscape has frozen. I'm running Inkscape 0.91 and Python 2.7.

TurnkeyTyranny, 2017-03-29:
Hi guys, unfortunately I have seen the same issue on another PC running inkscape 0.91 where the please wait box remains even though the gcode file has been created. I wasn't able to find the cause of this though. Seems very intermittent. I just make sure I save my svg before exporting it these days.

mentaldemise, 2017-06-28:
Happens to me too. Could be the Windows 10 Creator Update?

ExplodingLemur, 2017-06-28:
No, this was before the Creator Update.

frichsness, 2018-01-22:
i had this problem aswell, but i fixed it by using the zipped python27 files instead of installing python.

everything went fine with installing the python, except the last bit (easy_install lxml) that for some reason gives alot of errors (something about missing lxml files.