TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin #16: Negative Raster Images not processed correctly

gwrx, 2016-12-20:
There is a particular issue with raster images. "Normal" raster images (black on white) are cut perfectly, but inverted raster images (white on black) are not. I need to cut a stamp, and therefore I need to carve away the background, and only leave the parts for stamping. 
At the moment, some issue occurs when exporting: The black background gets cut, but the white parts of the image get cut too (and deeper). I tested with all the settings and tricks I know of - no change. Same problem occurs on Linux (Ubuntu) and Windows.

gwrx, 2016-12-21:
I solved it: Don't use RGB images for raster output. Convert the image to 1-channel-bitmap first! Then it works fine.

closed 2016-12-21 by gwrx

mkeyno, 2017-06-06:
@gwrx can you  explain how did you do that in inskscape .92 ? mine not created any G7 code

gwrx, 2017-06-08:
maybe there is no direct way in inkscape to convert an image to 1 bit - I think I did that actually in gimp

mkeyno, 2017-06-08:
have you ever successfully generate raster gcode (G7)with this plugin ?

gwrx, 2017-06-08:
yes

mkeyno, 2017-06-08:
can you please send me your setting you have used? I failed to create any raster code in inkscape , it only created the vector code, I don't know what did I miss

gwrx, 2017-06-09:
First of All: You need to ungroup everything, and you need to be sure nothing has any effects (shadows, rounded corners, clipping etc.). Your objects need to be objects with paths only.
For raster images: open your image in gimp, go to "Image > Mode > Indexed" and select "Use black and white (1-bit) palette", export as *.tif (since jpeg, png etc. don't support 1-bit color), import that tif in inkscape and export as gcode. You should see in the report-dialog that now one object is exported as raster...
If you want to engrave a path object (i.e. text), just go to "Edit > Make Bitmap Copy" in inkscape, and use this copy.

We are using this [firmware](https://github.com/timschmidt/buildlog-lasercutter-marlin) for our lasercutter with ramps