TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin #18: Text gives coordinate error

mentaldemise, 2017-06-28:
Every time I try to use Text it tells me "Unable to generate raster for object text123 as it does not have an x-y coordinate associated." -- Am I doing something wrong?

mayhem2408, 2017-06-28:
@mentaldemise Can you share with us a sample SVG file that is causing the problem?

mentaldemise, 2017-06-28:
I think I tracked it down. It's not text, but text that's transformed. If I just click and type it's okay. If I click and type and then drag the box to resize I get this issue.

mayhem2408, 2017-06-28:
@mentaldemise I just tried a new document, types 'Hello World'. Resized the text by dragging, changed the layer name and run the extension. Did not get an error and the Gcode looks fine.

mentaldemise, 2017-06-28:
Can you export this? Maybe I'm doing something wrong.
[textdemo.zip](https://github.com/TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin/files/1110186/textdemo.zip)

mayhem2408, 2017-06-29:
@mentaldemise Using the turnkey plugin I got the same error. I ungrouped all the objects and it worked OK. I regrouped the text at it worked. Not sure what the problem is with the file. But ungrouping and regrouping seems to have fixed it.
[textdemo-fixed.zip](https://github.com/TurnkeyTyranny/laser-gcode-exporter-inkscape-plugin/files/1111092/textdemo-fixed.zip)